Desuuuu/DGUS-reloaded #1: Custom G-Code Command Entry Using TFT Not Functioning

jlwainwright, 2020-06-21:
Printer: Creality CR10s-Pro V2

Firmware: https://github.com/Desuuuu/Marlin

System Board: Creality 3D v2.4

Issue: No Text Entry When Attempting To Type Custom G-Code Commands Into Pop-Up Keyboard.

Desuuuu, 2020-06-21:
This was happening at some point before because the input was filled with spaces when entering the screen.

Can you try pressing the backspace key a few times and try typing commands after that?

jlwainwright, 2020-06-21:
Tried backspacing multiple times and also back-arrow now and no luck with that. I tried powering the printer off and on just to make sure but still not working.

Desuuuu, 2020-06-21:
Thank you for the details, I'll investigate.

Desuuuu, 2020-06-23:
I wasn't able to reproduce this behaviour using the precompiled firmware and the DWIN_SET from the release.

I think the key difference between our setup is that my touchscreen is running a more recent version of the touchscreen's core firmware.

The version I'm using is available [here](https://github.com/Desuuuu/DGUS-reloaded/tree/master/extras/core_firmware).

A failed flash of these files could brick your touchscreen, so make sure not to cut the power to your printer during the flashing procedure, and consider unplugging any USB cable from the printer beforehand, for good measure.

If you want to try updating, they can be flashed using the same procedure as flashing the touchscreen:
* Create a DWIN_SET folder on your SD card
* Drop the 2 _.BIN_ files in this folder
* Insert the card into your touchscreen

jlwainwright, 2020-06-24:
This did not resolve the issue either. Although manual G-Code entry would be awesome from the TFT, the lack thereof is not a dealbreaker.

Desuuuu, 2020-06-24:
I understand the fact that it's not the most important feature but I'd still like to find out why that problem is happening and possibly resolve it.

I haven't been able to reproduce this behaviour with any of the 2 pre-compiled firmware files so far, so unless you've modified & compiled the firmware yourself, I'm led to believe this has something to do with the touchscreen firmware.

Now if these previous steps haven't fixed it, the only other possibility I see is that this is caused by something that was previously flashed to your screen (but that would surprise me).

In this case, you can reflash the screen using the `DWIN_SET.tar.gz` archive's content, with the `T5UID1.CFG` replaced by [this one](https://github.com/Desuuuu/DGUS-reloaded/tree/master/extras/flash_reset). This should make the screen erase its memory before the flashing process starts.

jlwainwright, 2020-06-24:
Hi @Desuuuu 

I have originally started off by using T5UID1.CFG at [github.com/Desuuuu/DGUS-reloaded/tree/master/extras/flash_reset](url)

For what it is worth, this is the details of the TFT that I have installed:

<blockquote class="imgur-embed-pub" lang="en" data-id="a/XEtHNPu"><a href="//imgur.com/a/XEtHNPu">TFT</a></blockquote>

Desuuuu, 2020-06-24:
Thanks for the picture. That's exactly the same model as mine.

Well, in this case I don't really have an idea what's causing your issue and I haven't found a way to reproduce it.

Desuuuu, 2020-06-26:
I went back and did some more testing. This is definitely caused by the touchscreen's core firmware being outdated, as I suspected earlier.

Updating the core firmware, as I described previously, does solve the issue.

jlwainwright, 2020-06-28:
Hi @Desuuuu

Flashing the TFT with [https://github.com/Desuuuu/DGUS-reloaded/releases/tag/1.0.1](url) has resolved the issue and I am now able to enter G-Code via the Display.

Thanks!

closed 2020-06-28 by jlwainwright