Desuuuu/DGUS-reloaded #3: [FR] CR-6 SE Support

randomgitname, 2020-08-18:
I've been attempting to convert this project to the new machine. I think the TFT in this machine is a rebranded DMG48270C043_03WTR. This is based on the T5L asic and uses DGUSII. 

For this UI to work on the new board, the project needs to be converted from .hmi to .dgus. dwin's dgusII 8.0 software attempts a conversion, but I didn't design the UI and have less than a few hours in this tool. I am unsure of how well it does the conversion.

I do know that the formats and containers have changed. Images are coverted to jpgs and placed into .ICL files. This saves about 11 MB on the flash which is nice. Sound files are placed into .wae containers. I completed these conversions.

One road block in the conversion process was .ICO files are now expected to be .ICL and i don't have anyway to extract the icons from files 24_icons 27_buttons 30_progress_left 37_progress_right.

If you placed the original files into folders in the project folder on github, finishing the image conversion could be completed.

I'm sure that there are other issues @ merlin, DWINOS_. Any feedback is welcome.

Desuuuu, 2020-08-19:
There's a way to extract BMP files from the ICO ones: by making DGUS Tools preview the image it will automatically extract it and save it as a BMP in the correct folder. But that's quite a tedious process so I added the BMP files in commit fbe4fd54626b4d0f059bdf80dac585700d29f905.

I haven't looked into the T5L screens at all, so I'm unaware of the differences in terms of how they operate. I guess the changes could be integrated into my fork of Marlin but I don't know if I'll find the time to look at this myself.

WadeRobot, 2020-09-16:
How is the progress of T5L? @randomgitname @Desuuuu

Desuuuu, 2020-09-16:
I don't have time to work on this and I don't have one of these screens on hand, so it will be up to someone else to make this happen.

randomgitname, 2020-09-17:
I don't know if mainline marlin supports the T5L with touch yet, Seems like alot of the work is going into the e3v2 currently which is T5 with rotary. Example marlin configurations supporting the cr-6 se have not been released. Something is holding them up or no one is working on it?  Idk if they are waiting for creality to release source code? I sent creality a request and this was the reply "About the source code ,due to commercial privacy, our R&D department does not release the source code of the firmware." a flip flop from the kickstarter page "We will open source the firmware after the crowd-funding. Please stay tune on our official SNS." 

So using desuuu's ui on these machines is on hold til some working open source firmware shows up. converting from DGUS to DGUS2 isnt the holdup.

closed 2020-11-23 by stale[bot]