Desuuuu/DGUS-reloaded #4: gcod console miss space touch

labotecno, 2020-09-11:
hello the gcod console miss space touch to make it work well , and how to get inside debug menu ?
thank you for this great firmware ! 
[image]

Desuuuu, 2020-09-11:
Since Marlin's G-code parser ignores spaces, I made the decision not to include a space key on the layout.

You can type your commands without spaces and it shouldn't cause any issue: `G0 X10 Y10` is equivalent to `G0X10Y10` for example.

The only case where I can see it being a problem is when passing a string, which is only the case with `M117` I believe, and it doesn't have much use in this situation.

For the debug menu, you need to press the screen version 10 times on the Information screen.

labotecno, 2020-09-11:
ok!!! i get inside debug menu ! thank you can you explain me how to get working gcod from dwin screen ? and i was see to modificate the firmware at level graphic , i want to implement G34 and bltouch move ... where is the file to manage the screen?
thank you !

Desuuuu, 2020-09-11:
You can modify the screen firmware with DGUS Tools. This will also require modifications to the mainboard firmware.

You'll have to learn on your own on how to make your modifications as I do not have time to teach you the details about this.

Here are two PDF that contain some documentation around the touchscreen:
* [DWIN DGUS DEV GUIDE_V40_2014.pdf](https://github.com/Desuuuu/DGUS-reloaded/files/5210854/DWIN.DGUS.DEV.GUIDE_V40_2014.pdf)
* [T5UID1.pdf](https://github.com/Desuuuu/DGUS-reloaded/files/5210855/T5UID1.pdf)

Good luck!

closed 2020-09-11 by Desuuuu

nicwilson58, 2021-08-11:
What about two commands at a time like  G28 G29,  will G28G29 work without a space? Marlin uses a semi-colon to addition of that might allow that to work

Desuuuu, 2021-08-11:
As far as I know there isn't any special character that will make this work without changing the implementation in Marlin.