Desuuuu/DGUS-reloaded #5: [Q] merging with klipper repo ?

dkuku, 2020-11-21:
Hi, I'm really glad that I found this repo when searching for klipper dgus.
I have an wanhao i3plus clone with an dgus lcd - its either 
    DMT48270M043_05WT
    DMT48270M043_02WT
depending on the clone
Im currently running the [advi3++]( https://github.com/andrivet/ADVi3pp) i firmware and I was searching for a way to get the display working with klipper - then I found your repo. 
I see that you have 2 separate repos for the lcd one for klipper and the other for marlin.
Do you think it is possible to join them and have one that can be used between both of them ? I want to have a way to test new firmwares but not flashing the lcd everytime and it seems that the repos are very similiar
keep up the good work

Desuuuu, 2020-11-21:
The two screen firmwares are indeed very similar, but there are a few differences between the two. Off the top of my head, the Klipper one doesn't include SD support or custom G-code input (that consequently led to a few design changes).

With that said, I kept the same addressing for the screen memory (you can have a look at these files for reference: [1](https://github.com/Desuuuu/klipper/blob/master/klippy/extras/t5uid1/dgus_reloaded/vars_in.cfg), [2](https://github.com/Desuuuu/klipper/blob/master/klippy/extras/t5uid1/dgus_reloaded/vars_out.cfg) and [3](https://github.com/Desuuuu/Marlin/blob/2.0.x/Marlin/src/lcd/extui/lib/dgus_reloaded/config/DGUS_Addr.h)), which means that you **should** be able to use them interchangeably.

I haven't tested that, but using the Marlin version of the screen firmware should mostly work fine, except for the missing features when running Klipper (which should do nothing). If you want to try, make sure you pay attention to what the printer does when using the screen controls and be ready to cut the power if necessary.

dkuku, 2020-11-23:
thanks - i have some problems with my extruder temp but when I solve this Ill have a look on that

Desuuuu, 2020-11-24:
I really don't think your touchscreen is compatible with this firmware. It's made for T5UID1-based models, like on the CR-10S Pro and it appears your model is different.

The firmware you linked also seems to be using a different format, which most likely confirms that it's not compatible.

dkuku, 2020-11-24:
Ok thanks for the info. I just got my printer running after few days - had
some problems, I need to use it a bit before I brick it again :)
-- 
Sincerely
Daniel Kukuła

closed 2021-01-13 by Desuuuu