Desuuuu/DGUS-reloaded #7: [FR] Light button

mrv96, 2021-01-06:
Hi, i would like to add a light button to your screen fw in order to enable/disable printer's led. Could you either add it or share the css of your bmp images?

I tried to do it myself, but there are slight gradients that i am not able reproduce.

Desuuuu, 2021-01-06:
The project is pretty much paused for now until I have some more time to work on it, at which point I will probably refactor it into some kind of flexible menu system (akin to Marlin's default menu system).

There's no CSS involved here (we're working with images) so I can't give you CSS, but I can give you the details of the buttons style:
* Background: Linear gradient, `#1f1f1f` to `#252525`, 130° angle
* Outline: 1px, `#111111`, 80% opacity
* Font: Calibri, 18pt, `#aaaaaa`

mrv96, 2021-01-06:
Thank you very much for the answer. I'll try to implement this feature and make a PR. I'm glad to hear that you will make a flexible menu system. It would be very appreciated.

In the last marlin bugfix there is a lot of new DGUS development. I hope that soon or later, your UI will be integrated in official Marlin, because it is one of the best available UI

closed 2021-03-15 by stale[bot]