Desuuuu/DGUS-reloaded #10: Weird Screen

Erik262, 2021-01-29:
Hi folks,

does someone know what's the problem here is??
[image]

Desuuuu, 2021-01-29:
What's the model of your touchscreen? If you can't tell, a picture of the back of the screen would be enough as well.

Erik262, 2021-01-29:
Of course...
[image]

Desuuuu, 2021-01-29:
Your touchscreen isn't compatible with this firmware.

It's a 320*480 screen while this firmware is for 480*272 touchscreens (it wouldn't even work in landscape due to the different resolution).

closed 2021-01-29 by Desuuuu

Erik262, 2021-01-29:
So how can I solve this? Do I have to write a completely new firmware or can I use yours and modify this a bit?

Desuuuu, 2021-01-29:
Unless you want to rewrite the whole thing, there's no way for you to make this work with your screen.