Desuuuu/DGUS-reloaded #11: Support for Mini DGUS LCD-s in progress

nepeee, 2021-03-14:
Hi!

I have a Wanhao Duplicator i3 Plus printer and i recently bought a BTT SKR MINI E3 V2.0 main board to upgrade to TMC drivers.
This printer has a DMT48270M043_05WT so i searched for some solution to use it with the new board and i found this project.
I have some experience with these type of displays, wrote some code in the past for them so i made some efforts to make the code work. It seams like not much changes required to make it happen, i done some initial test and for first glance everything seams to be working.

Changes i made to get it working:
- Changed the SwitchScreen function (the Mini DGUS lcd-s needs different packet)
- Removed the voice id from every button (the Mini DGUS lcd-s only have a basic tweeter so without this modification the buttons doesn't worked)
- Adjusted the Text Display font XY size (with the original config the text was extremely large)
- Replaced the 0_DWIN_ASC.HZK font file (with the original no text was displayed)

I have some question:
Can we work together to integrate my changes to this project or do you prefer to go with my own way?
Why does this project not part of the Marlin source like the "DGUS Touch Display with DWIN OS." module?

I have some other stuff that can extend this project if we work together:
I previously wrote some html/js code to allow the user to change the backgrounds/icons and border colors. For the user it's super simple to use, just select a background image, adjust the text button colors and the changes updated in real time. You can also export the bitmaps as a zip file to replace the original images.  I stopped working on it because i lost interest but i like to continue with it. The screen layouts was written in HTML and styled with CSS. Here's a quick demo:
https://www.youtube.com/watch?v=YRzkOEi-Ybo

I think i can make this work with your project.

Desuuuu, 2021-03-14:
I'm in the process of refactoring most of the firmware, so there's not much point submitting changes for now.

When I'm done with that, I will probably look into adding support for more touchscreens.

I haven't submitted a PR to Marlin because I'm not completely satisfied with the current state of the project. I will also reconsider that at a later point.

Regarding your screen tool, I don't really have a use for it. However, if you want to go further with that, I would suggest implementing features from DGUS Tool, which is quite buggy in my experience. An alternative to DGUS Tool would be great.

nepeee, 2021-03-14:
Unfortunately as far as i know there is no information about the data structure of the files generated by the DGUS tool :(

Desuuuu, 2021-03-14:
For the controls (13.bin and 14.bin), the format is pretty simple, it's a binary description of each control and is documented in the DGUS Application Guide PDF (I can post it later if you want).

The 22.bin file is used to initialize the display RAM, so it's just the data with a special format (there's an offset and it's "shifted" due to the screen addressing).

I assume the format for the icons isn't very complicated even though I haven't looked too much into it. It shouldn't be too hard to figure out how the pictures are arranged.

I haven't looked at the font files format either, I don't really know what to expect there.

nepeee, 2021-03-16:
Thanks, it looks promising. I will make some test code to confirm that we can work with the binary config files. It will be nice to generate the C header files so wee can use it directly. A VP address usage map would be also useful.

My lcd:
Based on my tests it's very similar to the T5UID1 displays.
The screen selection is different(maybe my version is compatible with the T5UID1):
`WriteHeader(0x0300, 0x80, 1); LCD_SERIAL.write((uint8_t)screen);`

For the font sizes i confirmed that i need to halve the pixel XY values to make the texts fit correctly. 
I done more testing on my printer, everything working except the screen brightness/volume menus. I haven't tested the G code menu yet.

Desuuuu, 2021-03-16:
I wouldn't bother with the header files at the moment, or anything related to the mainboard firmware really. Most if not all that stuff is going to be refactored.

From what I can tell, the system interface for your screen is different (RAM addresses 0x00 - 0x0FFF on T5UID1). Anything using this (screen switching, brightness/volume control, playing sounds and enabling/disabling touch controls) will most likely not work. It could also do something it's not supposed to, hard to tell without knowing how your screen works.

Anyway, I wouldn't recommend using this firmware until we know what we're daeling with.

nepeee, 2021-03-16:
It's better than no display at all :)

Desuuuu, 2021-03-16:
There's an alternative firmware for your printer [here](https://github.com/andrivet/ADVi3pp) with its own LCD implementation.

The documentation for your touchscreen is also available in the repository.

nepeee, 2021-03-16:
Yes i know it's based on my previous work. Its not compatible with the BTT SKR MINI E3 V2.0 ARM board as far as i know.

nepeee, 2021-03-16:
The touch data parsing is working, long way to go but a good start:
[image]

Desuuuu, 2021-03-17:
Looks promising!

closed 2021-08-21 by Desuuuu